isaac879/Pan-Tilt-Mount #1: Sourcing parts

TaylorHokanson, 2020-09-28:
Hi Isaac! I'm planning on building one of these, but I'm having trouble with some of the part specifications. Could you be more descriptive about the motors, for example? Are they uni/bi-polar, what is the torque value, etc.? I'm also curious to hear about which suppliers you used, as Digikey (I'm in the states) doesn't recognize the Bluetooth module part number among others. How long should the belts be? Thanks!

MrScottFletcher, 2020-12-23:
Amazon.com had all of the parts in December 2020.  Check the "3-Axis Slider Parts List.pdf" for part numbers.

jonowelser, 2020-12-28:
I've also been hitting speedbumps in the process of sourcing parts, and I can see descriptions/dimensions but not specific part numbers in the "3-Axis Slider Parts List.pdf".

This may help with the motors:

- **For the 42 x 42 x 23mm stepper motors**: I was able to see the motor data tag [at 2:30 in this video](https://youtu.be/1FfB7cLkUyQ), and it looks like the manufacturer is "Two Trees" and the model is "17HS4023". If so, the exact motor [may be on Amazon here](https://www.amazon.com/Twotrees-Nema-17-Stepper-Motor/dp/B07ZVLYL25).
- **For the 42 mm x 42 mm x 40 mm stepper motor**: Its harder to see the tag in the clip above, but the part number may be "SL42STH40-1684A" (no manufacturer visible). If so, [here are motors specs](https://3dprint.wiki/reprap/anet/a8/steppermotor) and the manufacturer may be Guangzhou Shenglong Motor Co., Ltd - I can't find the exact model on their AliBaba page, but found two other listings on AliExpress for this part number which can be found [here](https://www.aliexpress.com/item/32842920299.html) and [here](https://www.aliexpress.com/item/32722953463.html). I'll probably source a motor with the same specs from somewhere else ([possibly like this](https://www.amazon.com/DROK-2-Phrase-Universal-Electric-Engraving/dp/B06XSYP24P/)). 

Here where I'm stuck: What is a stepper motor driver I can use? 

- **Assuming TMC2208 is correct (see below)**: Does the "version" matter? Ex: [this listing](https://www.amazon.com/Printer-TMC2208-Screwdriver-Controller-Ramps1-4/dp/B082LSQWZF) on Amazon is for TMC2208 V1.2, but [this one](https://www.amazon.com/BIGTREETECH-TMC2208-Stepper-Driver-Printer/dp/B07WGCF3L2) is for TMC2208 V3.0 - does that matter at all? Also, that second listing also has different models for STEP and UART modes; which mode would likely be better?
- **TMC2208 vs A4988**: I noticed that parts list describes the stepper motor drivers as part number TMC2208, but the wiring diagram ("Schematic_Pan Tilt Camera Mount_Sheet_1_20200401234411.png") lists them as A4988, which is also what they look like in [this video](https://www.youtube.com/watch?v=uJO7mv4-0PY) from May 2020. They mostly* seem to have the same pinouts, but the TMC2208's seem to have taller heatsinks than the A4988's, although I haven't gotten far enough to see if that's an issue (I just know the final assembly is nice and snug). 
  - *While searching, I think I saw a couple TMC2208's and/or A4988's that had the outputs 1A / 1B / 2A / 2B  in a different order than on the schematic / other manufacturers on the market (just make sure the order is 2B / 2A / 1A / 1B like the schematic, reading from top to bottom).

PS I just want to say this whole project is absolutely incredible. The level of quality and attention to detail in everything from the the 3d files, code, YouTube videos, and documentation in this repository are amazing - this is one of the best 3d printing/electronics projects I've ever seen shared, and I can't wait until I'm done building it!