isaac879/Pan-Tilt-Mount #4: STP sources for several STL files are missing on Thingiverse

Bougakov, 2020-10-24:
Hi. I think the slider carriage files are missing from the archive with the STP sources. Could you please upload all the files here? STP files are ASCII-based, so GitHub is a perfect place for them.

Bougakov, 2020-10-24:
Oh, I sorted this out - the slider carriage files were in the earlier iteration. Created a PR that mentions it in README

Bougakov, 2020-10-24:
Oops, no. Those older files are for 2020 profile; STP files for 2040 are nowhere to be found :((

isaac879, 2020-10-24:
I've just added all the STEP files to the GitHub repository. I think there might have been some issue with thingiverse because someone else was also missing a files but when I check the zip folder I uploaded they were all in there... Not really sure what caused it but anyway the files are here now too.

Bougakov, 2020-10-24:
Thanks. I have two parallel 2020 bars (they can be assembled in 20*80, or 20*100 basically using shorter cuts); I will try to figure out how to tinker with the carriage to make the distance between wheels larger so it would fit my rails:

[image]

dronecz, 2021-03-07:
Hi, as I´m doing some modifications to you design, I download you GH files and there is STEP file missing for "Pan Mount" file. There is one in STL format, but that is terrible format for editing. Can you @isaac879  pls upload it?