isaac879/Pan-Tilt-Mount #20: Pinout

Cysign, 2021-11-04:
Hi,
I'm not sure what to connect to the different pinheaders.
Is there a building tutorial (like on instructables)?

H1, H2, H6, H7, P2 are not clear for me.
H should be the hall sensors. But which one belongs to which place?
And what's P2 good for? Is it the camera trigger?

jkowalk, 2021-11-05:
Have you had a look at the Schematic_Pan Tilt Camera Mount_Sheet_1_20200401234411.png. Also there are Videos on the YouTube channel isaac879. Also you can check the Arduino code to see which pins are used for what.

H2, H6 and H7 are connected to the hall sensors.
H1, I think, is not used in the current version without a programmable led.
And P2 is for the trigger.

I hope that helps.

Cysign, 2021-11-05:
That's why I'm asking. There's no information about the connected components for the P2, H1, H2...
I didn't have a look at the code yet, as I'm still printing parts ;)
Youtubevideos are never a good reference for schematics... they might explain steps for beginners....but I prefer the traditional way with pictures, legends and schematics.
At least most of the videos about electronics don't contain any information about revisions and pinouts...