isaac879/Pan-Tilt-Mount #23: Loop

Nikitore, 2022-03-10:
I can't get it to work. I am sending the commands through the serial port of the Arduino Nano but the motors keep looping back and forth. Do you know why this is happening? I send a video where you can see the problem.

Nikitore, 2022-03-10:
Video 

https://drive.google.com/file/d/13yk88zgiTplqRZulqzlmxvYA5bjrUprQ/view?usp=drivesdk

jkowalk, 2022-03-11:
Hi,
A quick google search helped me find these tips:
- Check if Speed is too high
- Check wiring (you might need to swap two wires; check out [this guide](https://buildbotics.com/wiring-stepper-motors/))
- Correctly adjust the potentiometer of the motor driver ([maybe this helps](https://all3dp.com/2/vref-calculator-tmc2209-tmc2208-a4988/))

Hope it helps...

Nikitore, 2022-03-11:
Thanks man! I'll check it out and let you know if it works.

Nikitore, 2022-03-23:
Hi! @jkowalk I checked the wiring and that was it. It was connected wrong. Thanks for your help.

closed 2022-03-23 by Nikitore