isaac879/Pan-Tilt-Mount #25: The slider doesn't change direction when I tell it to.

Nikitore, 2022-04-09:
Hi, as I show in the video, when I try to change the direction of movement of the slider with the function ¨j¨ this does not change and continues in the same direction. 

https://user-images.githubusercontent.com/99223586/162585901-b715d1ce-508c-41ad-9c05-24d15e7a3f28.mp4

I have already tested the wiring of the motors, changed the TCM 2208 driver, the continuity of the board between the output 3 of the arduino and the input inv of the driver. Everything indicates that this would be fine. But the problem continues, I am inclined to think that the Arduino is not sending the electric pulse to the driver to change direction.

Do you have any idea what it could be? 

Translated with www.DeepL.com/Translator (free version)

isaac879, 2022-04-11:
The Arduino or TMC2208 might have a damaged/not working pin. I'd try replacing the Arduino and motor driver and see if either of them fixes the problem. 
Does the slider move in the opposite direction if you enter a negative value?

Nikitore, 2022-04-11:
If I set negative values it keeps moving in the same direction. I tried swapping the stepper motor drivers and they all work correctly with the pan and tilt motors, but none of them work with the slider. So I was inclined to think that it might be the Arduino pin is damaged. I will try with another Arduino and let you know how it went.

Nikitore, 2022-04-15:
Hi, I have already changed the Arduino and now the slider axis is working, but unfortunately the pan and tilt axes stopped working. What should be the ideal Vref in the drivers? I have the following motor models. Pan and Tilt Usongshine US-17HS4023 and Slider KS 2sth40-1204a. In all 3 cases the Vref is set at 0.92 volts. Could this voltage be affecting the correct operation of the motors?

Nikitore, 2022-04-15:
And I think there must be some short circuit because both pan and tilt motors only being connected drained the battery very fast and overheated.